rogy-AquaLab/sinsei_UMIUSI_control #18: 構成決め

Chikurin1014, 2025-05-11:
- [ ] ROS2 controlについて理解する
- [ ] Hardware ComponentsとState/Command Interfaceの構成を決める
- [ ] Controllerの構成を決める

yk4to, 2025-05-21:
暫定の構成図です

[image]

[image]

Chikurin1014, 2025-05-21:
各コントローラーの役割が知りたいです。特に、`Gate`と`CAN manager`が気になってます。

yk4to, 2025-05-21:
以下のような役割を想定しています。
Interfaceを単純化するためにControllerはなるべく減らした方がいいんでしょうか？
 
- Gate
  - ros2_control外からのトピックを受け取る末端のコントローラー
- App
  - IMUの値を用いた姿勢制御を実装する部分
- Thruster（×4）
  - BLDCとサーボへの指令を受け取って振り分ける
- Indicators（いらないかも）
  - GPIO直結のLEDと、CAN経由のLEDテープへStateを振り分ける
- Cam（いらないかも）
  - 2つのカメラの入出力をまとめる？
- CAN Manager
  - 4つの基板へのCAN通信を一つのバスで行うため、入出力を振り分ける

Chikurin1014, 2025-05-21:
個人的にパッと思ったのは、

- `CAN`の中でも通信する相手によって頻度や優先度が変わるはず
  -> `CAN`は分離できないか
- `CAN`の中にスラスタの`joint`の情報を保持するため、`CAN`内で信号のエンコーディングをしなくてはならない
  -> `CAN Manager`と`CAN`はくっつけられないか
- スラスターに関する情報が`Thruster`で完結していない
  -> スラスター周りのセンサー値を`Thruster`から返すようにできないか

ですね

Chikurin1014, 2025-05-21:
`Cam`は多分いらないかなと思います。2つのカメラ映像は、結局別々のトピックに載せて発信することになるだろうし。

`Indicators`についても、受信するトピックをもとに判断すればよいと思います。
受信するトピックが表示するLEDと常に対応しているのであれば`Indicators`は不要ですが、対応していない場合がありうる(1つのトピックで内部状態に関する情報をすべて受信する場合など)は`Indicators`をかませる必要がありそうです。

Chikurin1014, 2025-05-22:
新しい図の方こっちでも見られますか？

yk4to, 2025-05-22:
新しい図を送りますね。

色々考え直してControllerはかなり簡略化できました。

[image]

Figmaのリンクも載せておきます。

https://www.figma.com/board/DVDbni3iVSgTYfv5mCJyMW/sinsei_umiusi_control?node-id=0-1&t=oFAYfpTHMWNYeMSz-1

- CAN通信の振り分けは`CAN`のHardware Component内で行うことにして`CAN Manager`をなくしました。
- スラスター周りのセンサー値を`Thruster`から返すようにしました。
- `Vector`の中身は速度`(Vx, Vy, Vz)`とオイラー角`(θx, θy, θz)`です。（適当に書いてしまっていました…）
- `Cam`は削除しました。
- `Indicators`も、そもそも基板上にインジケーターが1つしかないようだったので削除しました。

Chikurin1014, 2025-05-23:
ずいぶん良さそうです👍

Chikurin1014, 2025-05-23:
あと、IMUから`Velocity`も取りたいかな

Chikurin1014, 2025-05-23:
改めて https://github.com/rogy-AquaLab/sinsei-UMIUSI/issues/59 と照らし合わせて問題なさそうならこのIssueを閉じて、適宜Issueを立てつつ作業を始めてください。

yk4to, 2025-05-23:
> スラスターのトルクon/off命令はあって良いかも

確かにそうですね。
付け足しておきます。

> あと、IMUから`Velocity`も取りたいかな

ここで言う`Velocity`は角速度と速度どちらでしょうか…？

図の左端で受け取っている`TargetVelocity`は速度のつもりでした。

Chikurin1014, 2025-05-23:
速度のつもりです。

yk4to, 2025-05-23:
BNO055から取得した加速度を積分することで速度を得るということですか？

Chikurin1014, 2025-05-23:
BNO055って内部で速度計算してくれないんだっけ？

yk4to, 2025-05-23:
僕が調べた限りだと、内部で計算してくれるのはオイラー角とクォータニオンだけで速度は計算してくれないみたいです。

精度は落ちそうですが一応積分したものを取れるようにしておきます。

Chikurin1014, 2025-05-23:
あとは問題なさそうに見えます。チェックリストを確認して図を更新したらIssueを閉じてもらって大丈夫だと思います。

yk4to, 2025-05-23:
[image]

Chikurinさんのアドバイスをもとに修正しました。
あとチェックリストを見たところ浸水センサーを忘れていたので追加してあります。

色々とありがとうございました。
このissueはクローズしますね！

closed 2025-05-23 by yk4to