rogy-AquaLab/sinsei_UMIUSI_control #174: Hardware Componentをasyncで起動することを検討

yk4to, 2025-08-04:


Chikurin1014, 2025-08-24:
@yk4to 試してみてもらえますか？

yk4to, 2025-08-25:
公式ドキュメント: https://control.ros.org/jazzy/doc/ros2_control/hardware_interface/doc/asynchronous_components.html

Chikurin1014, 2025-08-31:
どちらにせよ100Hzは出せないということ…？

Chikurin1014, 2025-08-31:
少しずつ周波数を落としてWARNINGが出ない箇所を探してみてもらえますか？`async`を使う場合とそうでない場合両方について知りたいです

yk4to, 2025-11-12:
asyncなしの100Hzでこんな感じ

```
[ros2_control_node-1] [WARN] [1762923532.144888098] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 100 Hz. The loop
 took 13.293660 ms (missed cycles : 2).
[ros2_control_node-1] [WARN] [1762923532.171816601] [controller_manager]: Overrun might occur, Total time : 10109.430 us (Expected < 10000.000 us) --> Read ti
me : 6271.191 us, Update time : 91.350 us, Write time : 3746.889 us
[ros2_control_node-1] [WARN] [1762923533.273451900] [controller_manager]: Overrun might occur, Total time : 11573.659 us (Expected < 10000.000 us) --> Read ti
me : 9059.264 us, Update time : 117.183 us, Write time : 2397.212 us
[ros2_control_node-1] [WARN] [1762923533.273696173] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 100 Hz. The loop
 took 12.095976 ms (missed cycles : 2).
[ros2_control_node-1] [WARN] [1762923534.283590093] [controller_manager]: Overrun might occur, Total time : 11871.562 us (Expected < 10000.000 us) --> Read ti
me : 8834.677 us, Update time : 470.732 us, Write time : 2566.153 us
[ros2_control_node-1] [WARN] [1762923534.283901031] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 100 Hz. The loop
 took 12.302056 ms (missed cycles : 2).
```

closed 2025-11-13 by yk4to