rogy-AquaLab/sinsei_UMIUSI_control #180: サーボがカクつく

Chikurin1014, 2025-08-07:
サーボモータが角度変化時に振動します。

yk4to, 2025-08-22:
VESC Toolで本番用のサーボを回すと振動しませんが、`/cmd/direct/thruster_controller_**/angle`のtopicを送りラズパイからCAN経由で動かすと振動します

| 要素 | 可能性 |
| - | - |
| サーボ | OK |
| VESCファーム | OK |
| lispBM（VESC内でCANを受信してサーボを動かす追加実装） | ? |
| `sinsei_umiusi_control` | ? |

yk4to, 2025-08-22:
`ros2 topic pub /cmd/direct/thruster_controller_rf/angle std_msgs/msg/Float64 "{data: 45.0}"`を実行し、VESC側で受信した値を確認したところ`0`と`2500`（=45度？）が交互に来ていたそうです

`sinsei_umiusi_control`に問題がある可能性が高そう？

yk4to, 2025-08-22:
[image]

Chikurin1014, 2025-08-22:
> `sinsei_umiusi_control`のどこかが原因なのは確定でした。
> 
> `can_model`で`thruster_angle`を（`main_power_enabled`と同様に）変更時のみフレームを送るよう変更したら直ったので、 #153 での変更が怪しいかもしれないです

これについて、単にCANが送られる頻度が落ちて安定しているように見えただけな気がしてる

yk4to, 2025-08-24:
大丈夫そうなのでcloseします